acmerobotics/ftc-dashboard #7: Unable to View & Edit Arrays (1 or 2 Dimensional) in Configuration Variables

sbhatnag02, 2019-08-21:
In the configuration variables menu, I am unable to view or change any arrays (both 1 & 2 dimensional arrays). All that shows up is the name of the variable, but it appears that the values cannot be displayed. 

Perhaps viewing arrays isn't supported, but that would be a cool feature to add.

rbrott, 2019-08-22:
> Perhaps viewing arrays isn't supported, but that would be a cool feature to add.

Yes, arrays are not currently supported. I couldn't really think of a compelling use for array configuration variables, although I'm interested to hear your use case (especially for multidimensional arrays). I'm most concerned about creating a general-purpose array viewer/editor that scales to multiple dimensions and various sizes.

sbhatnag02, 2019-08-23:
During auto, our motions are defined by passing in a desired coordinate on the field along with other parameters (motor power, robot orientation, etc). Usually we only have 4-5 parameters, but the number of rows can be pretty large depending upon the number of motions. Traditionally, we've stored this information in a 2D array, or a text files that is later read into an array. 

If possible in the FTC Dashboard, we thought it would be really useful to be able to edit the coordinates and other motion parameters in real time to speed up debugging/testing.

adamrubinfeld, 2020-08-26:
> Interesting use case. I'd be happy to merge an array PR, although I don't plan on implementing array support myself in the near future.

I have an idea how to so it contact me if you want to hear

D3rpyDolphin, 2021-02-08:
A valid use for configurable arrays, although quite niche, is for OpenCv Scalars since the parameter stored in its class is an array. You could just make vars for each value in the scalar but that gets annoying.

rbrott, 2023-11-05:
Fixed-size arrays are supported after #144.

closed 2023-11-05 by rbrott