acmerobotics/ftc-dashboard #8: error on luanch

David10238, 2019-08-22:
I'm using sdk 5.1, and I get this error when I launch the robot controller app.
https://gist.github.com/David10238/30e27f712cbee90104be786c74589aab
the line in the activity is FTCDashboard.start()

rbrott, 2019-08-23:
I'm not sure this is a v5.1 issue. I would try cleaning and rebuilding.

David10238, 2019-08-23:
Tried cleaning, same error. I'm downloading a fresh copy of the sdk to try without any additional edits

David10238, 2019-08-23:
so I found out it was one of the times I used the @Config annotation, I'll start adding the annotation back to find out where it caused the issue.

David10238, 2019-08-23:
so I did some more testing. it turns out Dashboard doesn't like lateinit
https://gist.github.com/David10238/8359dc825a18433a3e38cc4f9857f1d2
the Globals object throws an error whenever @Config is used. I did some testing in the crashtest object and it appears public lateinit variables throw an error, but not nullables or private lateinits. idk if there is any possible way of working around this on the dash end, but I am able to work around it on the user side.

rbrott, 2019-08-23:
Ah yes that makes perfect sense. `lateinit` is equivalent to an uninitialized value which is problematic for objects (`null` values are not allowed at any time). I don't really intend on changing this behavior (besides a better error message).