acmerobotics/ftc-dashboard #11: Dashboard doesn't load when Robot Controller is started by plugging in USB.

EddieDL, 2019-10-27:
If you happen to have the Robot Controller app shut down and plug in the robot hardware to the phone, the Robot Controller app "auto starts" but the Dashboard is not started properly. You have to leave the robot plugged in and restart the Robot Controller app.

rbrott, 2019-10-29:
What does "start properly" mean? Is the entire dashboard inaccessible (i.e., no web page/connection) or merely a subset of functionality (e.g., op mode control)?

EddieDL, 2019-10-29:
Sorry I wasn't clear. 

I forgot to test on SDK 5.3 last night, but in the previous SDK it just didn't start (the blue text never displays showing "Dashboard enabled"). When you attempt to use the dashboard you get a null reference exception. The text and exception type are from memory, I will explicitly test tonight.

rbrott, 2019-11-07:
I was able to reproduce this myself; there are some strange things going on with the root activity when the app is launched from a device attachment. Thanks for the report.

rbrott, 2019-11-09:
I have a lightly tested solution for you to try. Move the following lines in `FtcRobotController/src/main/AndroidManifest.xml` from the `FtcRobotControllerActivity` `<activity>` tag to the `PermissionValidatorWrapper` `<activity>` tag.
```
      <intent-filter>
        <category android:name="android.intent.category.LAUNCHER" />
        <action android:name="android.intent.action.MAIN" />
      </intent-filter>

      <intent-filter>
        <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
      </intent-filter>
```

EddieDL, 2019-11-13:
I should be able to test this out a bit tonight. Will let you know how it goes.

EddieDL, 2019-11-15:
I am not sure I really tested this how you intended. When I tried the change the following happens.

- Whenever I plug in the robot to the phone I get the dialog asking if we want to use the app when this device is connected. No matter how many times I say yes it keeps asking.
- I can't seem to get the hub to be recognized in this configuration.
- When the app is closed and I plug-in the app doesn't open.

Below is what I did.
```
    <activity android:name="org.firstinspires.ftc.robotcontroller.internal.PermissionValidatorWrapper"
              android:screenOrientation="fullUser"
              android:configChanges="orientation|screenSize"
              android:label="@string/app_name"
              android:launchMode="singleTask" >

      <intent-filter>
        <category android:name="android.intent.category.LAUNCHER" />
        <action android:name="android.intent.action.MAIN" />
      </intent-filter>

      <intent-filter>
        <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
      </intent-filter>

    </activity>
``` 
```
    <activity
      android:name="org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity"
      android:screenOrientation="fullUser"
      android:configChanges="orientation|screenSize"
      android:label="@string/app_name"
      android:launchMode="singleTask" >
      
      <meta-data
        android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
        android:resource="@xml/device_filter" />

      <!--org.firstinspires.main.entry indicates that this app is compatible with the Dragonboard Control Hub-->
      <meta-data
          android:name="org.firstinspires.main.entry"
          android:value="true" />
    </activity>
```

rbrott, 2019-11-15:
Could you move both USB-related `<meta-data>` tags to the permission activity?

EddieDL, 2019-11-15:
Will do. I only see one USB-related `<meta-data>` tag. The other seems to be related to `org.firstinspires.main.entry`.

NoahAndrews, 2019-11-15:
You can ignore the main.entry one. That honestly doesn't even need to be there anymore.

rbrott, 2019-11-15:
Sorry I meant for you to move both USB-related tags; you're correct that only one is a `<metadata>`.

EddieDL, 2019-11-15:
That makes sense. I wish I understood what all of this does more :( Guess I need to spend some time with the Android docs :)

EddieDL, 2019-11-15:
That seems to have done the trick. Now when I plug in the robot everything starts up and the dashboard appears to be working properly.

When doing this the Rev Hub isn't found until I manually restart the robot controller using the option menu in the app. My assumption is this doesn't have anything to do with the dashboard though.

closed 2019-11-15 by EddieDL

rbrott, 2019-11-15:
Yes, I think some of the intent-handling code should be moved to the permissions activity.... To be honest, I'm not sure whether I would use these modifications in competition.

EddieDL, 2019-11-16:
If I don't make the change, my fear is that an FTA will try to reset the phone and just plug it in. Then dashboard will not be started but the FTA will think it is good. Then we go to start an OpMode and it crashes and the only way to fix it is to make sure the app is started before the robot is plugged in.

The only alternative I can think of is to add logic everywhere dashboard is used to check if it is found. If not somehow revert back to using standard telemetry.

Is there another option you can think of?

rbrott, 2019-11-16:
I think the easiest potential solution is removing `FtcDashboard.stop()`. This might leak resources, but it should prevent the dashboard from stopping prematurely in this case.

rbrott, 2019-11-17:
I added a check for this startup condition to the stop method in the latest version. You shouldn't need to modify anything more from the usual now.

EddieDL, 2019-11-17:
That is awesome! I updated and that is working great. I still have issues where the Rev Hub isn't found but I am pretty sure that has to do with [this](https://github.com/FIRST-Tech-Challenge/SkyStone/issues/133)