acmerobotics/ftc-dashboard #12: App Crashes

qwertyquerty, 2019-11-07:
The dashboard has been causing our app to crash, mainly when hitting the back button to get out of settings or similar things.

rbrott, 2019-11-07:
Do you have logs/a stacktrace?

qwertyquerty, 2019-11-07:
I'll get you those as soon as I can.

One more detail, it seems that everyone that has been reporting crashes are using a moto g4. Might be a coincidence might not.

Flashman2000, 2019-11-24:
Having the same issue. I start up the robot control app and once I try to connect to the dashboard thru chrome I see "event loop detached"  and then restarting the app leads to app crash. On moto G5 Plus, android 8.1.0

rbrott, 2019-11-24:
Do you have a stacktrace? Phone model/OS isn't particularly helpful.

David10238, 2020-01-14:
I have a stack trace. I've also been able to reliably reproduce it by forcing the app to restart by getting stuck in stop. As soon as you completely close the app and relaunch it though it works perfectly fine. My theory is it has something to do with the fix for ftc_app issue #712 https://gist.github.com/David10238/23b1e1470678033ddfa80c8ae680cf61

closed 2020-08-28 by rbrott