acmerobotics/ftc-dashboard #14: Crashes on launch

nchowder, 2019-11-08:
The app crashes when opened. I'm getting this error:

    2019-11-08 18:01:55.970 3813-3813/com.qualcomm.ftcrobotcontroller E/AndroidRuntime: FATAL EXCEPTION: main
        Process: com.qualcomm.ftcrobotcontroller, PID: 3813
        java.lang.NullPointerException: Attempt to invoke interface method 'void com.qualcomm.robotcore.util.WebHandlerManager.register(java.lang.String, org.firstinspires.ftc.robotcore.internal.webserver.WebHandler)' on a null object reference
            at com.acmerobotics.dashboard.FtcDashboard.internalAttachWebServer(:517)
            at com.acmerobotics.dashboard.FtcDashboard.attachWebServer(:118)
            at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onServiceBind(:706)
            at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity$1.onServiceConnected(:193)
            at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1535)
            at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1563)
            at android.os.Handler.handleCallback(Handler.java:751)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:154)
            at android.app.ActivityThread.main(ActivityThread.java:6316)
            at java.lang.reflect.Method.invoke(Native Method)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:872)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:762)
    
Here is my `FtcRobotControllerActivity.java` file: https://gist.github.com/nchowder/e86a81e0ece858cf5b76ea5a3f767528

The line that throws the error is this:

    FtcDashboard.attachWebServer(service.getWebServer());

It appears that `service.getWebServer()` is null. Could you help me troubleshoot this?

rbrott, 2019-11-09:
Can you reproduce it?

rbrott, 2019-11-12:
Also are you using OpenRC?

nchowder, 2019-11-12:
Yes, I am, and I'm using a Moto G4. I will give you some steps to reproduce the issue soon.

NoahAndrews, 2019-11-12:
@nchowder I suspect you are using the Extreme Turbo build variant of OpenRC. That version doesn't include the webserver, so it is incompatible with ftc-dashboard. Make sure to switch to the normal Turbo build variant in Android Studio.

NoahAndrews, 2019-11-12:
@FROGbots-4634 You might want to consider mentioning that Extreme Turbo is incompatible with third-party libraries that use the webserver, such as ftc-dashboard.

rbrott, 2019-11-13:
You can still use the dashboard with Extreme Turbo if you really want. Just omit the `#attachWebServer()` call and run the dashboard locally using the development instructions (or host the generated static files under releases).

nchowder, 2019-11-14:
This solves my problem.

closed 2019-11-14 by nchowder