acmerobotics/ftc-dashboard #16: Vuforia SkyStone Nav & Camera Stream to Dashboard issues

ech03, 2019-12-03:
When using the ConceptVuforiaSkyStoneNavigation.java example as a foundation we are able to stream the phone camera to the dashboard using:  _**FtcDashboard.getInstance().startCameraStream(vuforia, 0);**_ if the last portion of the example file is removed, but if we leave it in we can't get it to stream to the dashboard. 

In other words...if u remove lines 309 - 359 (https://github.com/FIRST-Tech-Challenge/SkyStone/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptVuforiaSkyStoneNavigation.java) IT Works with just _**FtcDashboard.getInstance().startCameraStream(vuforia, 0);**_

If you leave lines 309-359 in and just add _**FtcDashboard.getInstance().startCameraStream(vuforia, 0);**_ it NO WORKY :(

rbrott, 2019-12-04:
I'm unable to reproduce this with my Control Hub and webcam. I can't think of a reason why the additional lines would cause any issues with the dashboard. 

Does the DS camera stream feature work in this case?

ech03, 2019-12-09:
I'm sorry long story short...more or less user error.

closed 2019-12-09 by ech03