acmerobotics/ftc-dashboard #20: Robot Controller App Infinite Crash loop

SamuelPCarter, 2020-02-12:
Currently running the latest road runner quick start release. I am having issues with the FTC Dashboard. The Robot Controller App infinitely restarts (crashes) when starting the code while connected to the Dashboard. The dashboard trys to reconnect and crashes each time it does so. Here is the Debug output:

Regards,
Sam Carter

[CrashLog.txt](https://github.com/acmerobotics/ftc-dashboard/files/4189346/CrashLog.txt)

rbrott, 2020-02-12:
`java.lang.IllegalArgumentException: Infinity is not a valid double value as per JSON specification. To override this behavior, use GsonBuilder.serializeSpecialFloatingPointValues() method.`

This is your problem. One of your configuration variables is infinity.

Jonathanzhao02, 2020-02-13:
I'm having the same issue. I'm using the road runner quickstart on app version 5.4, and the robot controller also crashes on start-up while connected to the dashboard. Do you know where I may be able to find the configuration?

Calvin-Xu, 2020-02-13:
I recommend commenting out `@Config` one by one on the classes until the issue disappears—then you know which class is causing the problem. Then comment out `public static` variables in the problematic class one by one until the issue disappears—then you know which configuration is causing the problem.

SamuelPCarter, 2020-02-14:
After commenting out @Config in DriveConstants.java, the dashboard worked properly. Thank you so much!

SamuelPCarter, 2020-02-14:
> After commenting out @config in DriveConstants.java, the dashboard worked properly. Thank you so much!

closed 2020-02-14 by SamuelPCarter

Jonathanzhao02, 2020-02-14:
Thank you for your help! I found I was inadvertantly dividing by zero at one point, causing my variable to return as infinite. Of course, I don't know if this is the same issue as what the original author was having.