acmerobotics/ftc-dashboard #23: page is empty

adamrubinfeld, 2020-07-25:
when i open the dashboard page and when i open the developer tools it print a error

react-dom.production.min.js:2549 Uncaught TypeError: Cannot read property 'content' of undefined
    at Object.getContent (LayoutPreset.jsx:99)
    at t.value (Dashboard.jsx:49)
    at yu (react-dom.production.min.js:3880)
    at vu (react-dom.production.min.js:3871)
    at Qa (react-dom.production.min.js:6116)
    at La (react-dom.production.min.js:5101)
    at Ua (react-dom.production.min.js:4958)
    at Ra (react-dom.production.min.js:4817)
    at react-dom.production.min.js:2543
    at t.unstable_runWithPriority (scheduler.production.min.js:338)

rbrott, 2020-07-26:
Seems like a local storage value was corrupted. Try clearing local storage.

adamrubinfeld, 2020-07-26:
how i do that?

adamrubinfeld, 2020-07-26:
work
tnx

closed 2020-07-26 by adamrubinfeld