acmerobotics/ftc-dashboard #26: Gradle sync failure with 0.3.9 and Open-RC Turbo

IronReignRobotics, 2020-09-10:
I'm getting Gradle sync errors when trying to upgrade from 
implementation 'com.acmerobotics.dashboard:dashboard:0.3.8'
to implementation 'com.acmerobotics.dashboard:dashboard:0.3.9'

This is using the latest Skystone 5.5 compatible release of Open-RC Turbo - the problem exists with all build variants:

Unable to resolve dependency for ':FtcRobotController@turboDebug/compileClasspath': Failed to transform artifact 'RobotCore.aar (org.firstinspires.ftc:RobotCore:5.5)' to match attributes {artifactType=jar}
Show Details
Affected Modules: FtcRobotController

The Show Details link doesn't actually show anything helpful, so I really have no idea what is going on.

Our version of the repo is available here: https://github.com/IronReign/OpenRC-Turbo

rbrott, 2020-09-10:
Are you sure this is caused by dash? I get that error with your repo with 0.3.8, and dash works fine in stock mode out of the box on a clean OpenRC clone (you may need to exclude transitive dependencies for other build variants).

IronReignRobotics, 2020-09-11:
that's strange. on my systems 0.3.8 works. then the only thing I do is change the last digit in the gradle implementation line from 8 to 9, tell it to sync gradle and then I start getting gradle errors.

i'll go with the theory that something got jacked up on our end. will try a rebuild starting from a clean OpenRC and let you know how it goes. maybe that will fix the gamepad issue too.

IronReignRobotics, 2020-09-11:
Continuing to work from our repo, I tried upgrading to gradle 4.0.1 and the sync failure went away. Now I get a build failure with a ton of messages like:

Duplicate class androidx.renderscript.Allocation found in modules androidx-rs.jar (androidx-rs.jar) and androidx-rs.jar (org.firstinspires.ftc:RobotCore:5.5)

So I went a pulled a fresh separate clone of Open-RC Turbo 5.5b. All I did was add the dashboard 0.3.9 implementation lines to the robotcontroller and teamcode gradle files, synced and built and I get the exact same Duplicate class errors. Note that on this clean 5.5b project dashboard 0.3.8 also has build errors - though they are different.

rbrott, 2020-09-12:
> So I went a pulled a fresh separate clone of Open-RC Turbo 5.5b. All I did was add the dashboard 0.3.9 implementation lines to the robotcontroller and teamcode gradle files, synced and built and I [...]

That's basically the procedure I followed. I ran `./gradlew assembleStock` successfully without modifications. The other variants build if you exclude the transitive SDK dependencies:
```groovy
implementation('com.acmerobotics.dashboard:dashboard:0.3.9') {
    exclude group: 'org.firstinspires.ftc'
}
```

IronReignRobotics, 2020-09-13:
Thank you! It was that exclude syntax that I needed. It's building now. I'm not running the Stock variant. The exclude syntax does appear to be compatible with the the stock build variant - at least it builds.

closed 2020-09-13 by IronReignRobotics