acmerobotics/ftc-dashboard #27: Gamepads not working with 0.3.8 and OpenRC Turbo

IronReignRobotics, 2020-09-10:
Gamepads connected to my development pc aren't registering - they are recognized by the system, but in dashboard they are not detected. Start A doesn't do anything. I am able to init/start/stop opmodes and use other features of dashboard. Running dashboard in chrome 84.0.4147.125

I have used gamepads with dashboard before - when running on a stock skystone 5.0 repo. This issue is with Open RC Turbo 5.4B. I'm using the gradle integration. Note that I did my own migration to AndroidX on Turbo 5.4B before the 5.5 stuff was released. Not sure if that is the problem. I've now upgraded to 5.5B but can't update to your latest - I filed as separate issue about that.

There isn't anything on your documentation site about troubleshooting joysticks, so submitting here.

NoahAndrews, 2020-09-10:
Are you using the same gamepads that worked before?

IronReignRobotics, 2020-09-10:
no I can't say that - we have a pile of the F310 gamepads and don't track them but are aggressive about dumping buggy ones. the gamepad works on a regular android driver station.

rbrott, 2020-09-10:
Here's an independent tool for testing HTML5 gamepad support: https://gamepad-tester.com/

IronReignRobotics, 2020-09-11:
thanks! gamepad-tester is great. the gamepads work there.

rbrott, 2020-09-11:
Do they still not work in dash?

IronReignRobotics, 2020-09-11:
Correct - they still don't work in dash

rbrott, 2020-09-12:
What's the gamepad's ID?

IronReignRobotics, 2020-09-13:
The gamepad is reported by gamepad tester as: Xbox 360 Controller (XInput STANDARD GAMEPAD)
I can have gamepadtester loaded up in another tab in chrome and the gamepad seems to be perfectly responsive

But in dashboard the gamepad icons stay greyed out. Start-A won't activate it.

Also I'm now running OpenRC Turbo 5.5b with dashboard 0.3.9

IronReignRobotics, 2020-09-13:
Device manager reports the last event as:
Driver Management concluded the process to install driver xusb22.inf_amd64_d80349221fa8b78c for Device Instance ID USB\VID_046D&PID_C21D\B489B9DF with the following status: 0x0.

rbrott, 2020-09-13:
Seems like the gamepad is probably being filtered out. The code is looking for `"Xbox 360 Wired Controller"` in the ID. I'm happy to change it to just `"Xbox 360"` for the next release.

commit d48e4bfd (2020-09-13): Broaden Xbox 360 gamepad criteria (fixes #27)
closed 2020-09-13 by rbrott