acmerobotics/ftc-dashboard #30: Stream camera view not with vuforia

adamrubinfeld, 2020-09-18:
Hi
I mentor team and contently we work on vision recognition and we need to test the pipeline and that very hard the see the camera view from robot control that any way we can send the camera view to the dashboard (we using easyOpenCV)

rbrott, 2020-09-18:
Yes, you can pass an instance of EOCV `OpenCvCamera` to dash `FtcDashboard#startCameraStream()`.

adamrubinfeld, 2020-09-19:
Can you send me example?

rbrott, 2020-09-19:
Take any EOCV example and just add `FtcDashboard.getInstance().startCameraStream(...);` with the appropriate arguments. For instance, take [this example](https://github.com/OpenFTC/EasyOpenCV/blob/master/examples/src/main/java/org/openftc/easyopencv/examples/InternalCamera1Example.java) and add `FtcDashboard.getInstance().startCameraStream(phoneCam, 0)` on line 98.

adamrubinfeld, 2020-09-21:
I will try tomorrow
I tried that before 4 days and that does a error of network or something

FROGbots-4634, 2020-09-27:
> I tried that before 4 days and that does a error of network or something

So..... yeah... about that.... due to an oversight on my part, the internal camera v1 implementation actually runs on the UI thread, and Android will crash your app on purpose if you do network on the UI thread. Try switching to the EOCV camera v2 implementation.

adamrubinfeld, 2020-09-28:
@FROGbots-4634 And if I working with webcam?

FROGbots-4634, 2020-09-28:
@adamrubinfeld webcams should not pose an issue.

adamrubinfeld, 2020-09-28:
Ok tnx
I will try tomorrow and give answer of that work

adamrubinfeld, 2020-09-29:
Work 
Tank you very much guys

closed 2020-10-01 by adamrubinfeld