acmerobotics/ftc-dashboard #31: Incorrect SDK dependency scope

FROGbots-4634, 2020-09-27:
A user on Discord reported that their build failed when using Dash with OpenRC Turbo 6.0A. Note that ExtremeTurbo now contains the webserver, so that isn't the issue.

The exact error they reported was:

```
Duplicate class androidx.renderscript.Allocation found in modules androidx-rs.jar (androidx-rs.jar) and androidx-rs.jar (org.firstinspires.ftc:RobotCore:5.5)
```

Upon review of your gradle scripts for the Dashboard module, I believe he was hitting a dependency clash as the result of incorrect scope used the gradle file. I think that the following lines need to be changed to `compileOnly` scope:

https://github.com/acmerobotics/ftc-dashboard/blob/1456730a6260a8495dc0d7734fc0b2c27b5814ba/FtcDashboard/build.gradle#L102-L107

FROGbots-4634, 2020-09-27:
Pulling down the [POM file from bintray](https://bintray.com/acmerobotics/maven/download_file?file_path=com%2Facmerobotics%2Fdashboard%2Fdashboard%2F0.3.10%2Fdashboard-0.3.10.pom), I can confirm that the SDK dependencies are leaking into the release build:

```xml
    <dependency>
      <groupId>org.firstinspires.ftc</groupId>
      <artifactId>RobotCore</artifactId>
      <version>5.5</version>
    </dependency>
    <dependency>
      <groupId>org.firstinspires.ftc</groupId>
      <artifactId>Hardware</artifactId>
      <version>5.5</version>
    </dependency>
    <dependency>
      <groupId>org.firstinspires.ftc</groupId>
      <artifactId>RobotServer</artifactId>
      <version>5.5</version>
    </dependency>
    <dependency>
      <groupId>org.firstinspires.ftc</groupId>
      <artifactId>FtcCommon</artifactId>
      <version>5.5</version>
    </dependency>
```

rbrott, 2020-09-27:
See https://github.com/acmerobotics/ftc-dashboard/issues/26#issuecomment-691387274

FROGbots-4634, 2020-09-28:
Well, that's certainly a bandaid fix that works, but I'm still of the opinion that transitive dependencies on specific SDK versions should not be leaked into the release build.

rbrott, 2020-09-28:
The main reason for including the transitive dependencies is enforcing a certain min SDK version. I understand that this causes issues for people with nonstandard SDKs, but I think it's the proper approach in general. I'm hesitant to adopt a different approach just for OpenFTC users.

FROGbots-4634, 2020-09-28:
Is it possible to check build config at runtime to determine the version? cc @NoahAndrews. In any case I don't actually understand why it *doesn't* explode on the stock SDK if it's not the same version it was built against. I would think that would still cause a dependency clash.

NoahAndrews, 2020-09-28:
Hmm, I think you could probably check the BuildConfig class in the relevant RobotCore package.

rbrott, 2020-09-28:
If I'm reading [the docs](https://docs.gradle.org/current/userguide/dependency_resolution.html#sub:resolution-strategy) correctly, Gradle will use the latest version when there is a conflict.

dgorbunov, 2020-11-08:
It would be good to add this fix in the docs for people that use both OpenRC and FTC-Dashboard - it seems like this will happen to anyone that uses both.

rbrott, 2020-11-10:
Good suggestion. The fix is now in the docs.

closed 2020-11-10 by rbrott

rbrott, 2025-08-17:
From @Windwoes:

> would it make more sense to check the BUILD_CONFIG file instead of the transitive dependency approach?

I'm open to a runtime check since the error messages for a compile-time mismatch are not very obvious.

> Because the current approach also means it's possible for teams to be mixing and matching library versions from different releases without knowing it.

I'm not super concerned about this. I'd be pretty surprised if mixed SDK library versions compile together but then misbehave at runtime. In any case it can be remedied by adding the full complement of SDK libraries to the POM.