acmerobotics/ftc-dashboard #43: Disable Dashboard should be accessible from DS

dgorbunov, 2021-01-14:
With the onset of Control Hubs, it is important to be able to disable the dashboard from the Driver Station phone.

TimVargo, 2021-01-14:
I was going to make a very similar comment.  Problem is that the DS code is not open source like the RC code is, so how could this be accomplished?  Perhaps a disable command could be built into the RC web interface, and/or an API call?

rbrott, 2021-01-14:
I see two easy options:
* Expose `FtcDashboard#enable()`/`FtcDashboard#disable()` (with changes to modify shared preferences) and let users incorporate them into their own op modes.
* Add instance op mode registrar that registers enable/disable op modes with `FtcDashboard#registerOpModes()`. 

I think I prefer the latter.

I guess disabling from the UI is convenient, but the enable/disable asymmetry bothers me.

closed 2021-01-20 by rbrott

dgorbunov, 2021-01-20:
Thanks for adding this. This could just be me, but upon bumping the version to 0.40 in build.gradle and syncing, it seems FtcDashboard.java isn't updated to include the new op mode registering. Is there something I'm doing wrong? I've tried invaliding gradle cache and deleting all of it.

rbrott, 2021-01-20:
I intend for registration to happen automatically without any changes. To suppress it, pass `true` to `FtcDashboard.start()`.

dgorbunov, 2021-01-20:
That method is not present in FtcDashboard for me, just start().

dgorbunov, 2021-01-20:
For some reason 0.3.10 is only visible to FtcRobotControllerActivity but in other classes I can access version 0.4.0

rbrott, 2021-01-20:
Ah you will have to update the version in both Gradle files.

dgorbunov, 2021-01-20:
Ah, totally missed that. Thanks!

TimVargo, 2021-01-22:
Nicely done!