acmerobotics/ftc-dashboard #45: Having trouble with FTC SDK 6.1

alan412, 2021-01-27:
Version FTC-Dashboard: 0.4.0
Version FTC SDK: 6.1

The dashboard comes up and says "Op mode controls have not initialized"

This is using Control Hub.

Our code is here if you want to try directly: https://github.com/ftc16072/UltimateGoal20-21/tree/openCV

rbrott, 2021-01-27:
Looks like you have the proper call. I suspect it's just a weird hardware attachment start that messes up the normal activity lifecycle. Does the issue persist after you restart?

alan412, 2021-01-27:
Yes, it persists after a restart.     Our electronics this year are pretty vanilla.   CH -> EH, a Logitech C270 off of USB, all standard motors and encoder wires.  some of the servos go through a servo power module, others are directly connected.   The only sensor connected is a 2M REV Distance sensor.   A REV Blinkin is on one of the servo ports.    We have a REV grounding strap and power is going through an XT30 expansion module.

I can try to go get our spare CH today and try it with nothing attached.    In the meantime, If you build our code and put it on your hardware, does it give you the list of opmodes?  (The configuration xml file is out of date so you won't be able to init an opmode, but if you get the list of opmodes then you are getting farther.)

I know this worked for us a while back on the CH ( we were using it to tune where our OpenCV is looking).   Unfortunately a TON of changes (both hardware and software including upgrading the SDK since then).    It looks like we rebased our commits on that branch so I am not sure I can find exact code that works.  (That was with 0.3.1.  We were getting a problem about opmode not attaching (sorry I don't have the exact wording) so we upgraded to 0.4.0)

rbrott, 2021-01-27:
> If you build our code and put it on your hardware, does it give you the list of opmodes?

Unfortunately I don't have any hardware at the moment.

>  (That was with 0.3.1. We were getting a problem about opmode not attaching (sorry I don't have the exact wording) so we upgraded to 0.4.0)

Makes sense. The 0.3.x "Event loop detached" error message was replaced by "Op mode controls have not initialized" in 0.4.0.

NoahAndrews, 2021-01-27:
Is your Expansion Hub connected to the Control Hub via USB or RS-485?

alan412, 2021-01-27:
RS-485.

alan412, 2021-01-28:
A few things I discovered last night (all on a Control Hub with nothing connected to it except power)
 * FTC Dashboard still fails - so not a hardware connection issue
 * Stock 6.1 works fine
 * I can find a version of our code that works - when we were on EasyOpenCV 1.4.1 
     * (Our most current was on 1.4.3)
     *  I tried moving our most current software to 1.4.1 - still didn't work 
     * Moving to the commit after the one that worked - sadly changed EasyOpenCV version and some code - did not
     * Changing the version of EasyOpenCV on that commit back to 1.4.1 still did not work
          *  I am a little concerned that maybe I needed to reboot the system here - this seems very strange

My current guess is that it is something in EasyOpenCV newer versions that requires FTC SDK 6.1 that breaks but I haven't been able to prove that yet.

alan412, 2021-01-29:
Ok.  I found the issue after a bunch of dead ends and it is in our code.   A student made a static variable inside a class that was annotated with @Config and was writing to it inside a class method.   Even though this class method was never called, this made FtcDashboard very upset and refuse to register OpModes.  (While we can quibble that the error message could be clearer, it is clearly a bug in our code.)

rbrott, 2021-01-29:
Huh I think might qualify as a bug. Could you link the offending variable in context?