acmerobotics/ftc-dashboard #48: Defaulting to standard gamepad mapping when controller is unknown

NoahBres, 2021-02-10:
I think this would require some testing with non-ftc gamepads but it would be neat to just default to the [standard gamepad layout](https://w3c.github.io/gamepad/#remapping) when the ID isn't known, enabling the use of non-FTC legal gamepads for use at home or something (there have been a non-zero amount of such requests, primarily Dualshock 5 and Xbox One controllers but not really important enough to spend significant time on imo).
With PR #47, the Dualshock 4 and Xbox 360 seem to have identical button layouts. I'm assuming that most modern gamepads will share the standard w3c spec.

jasonappah, 2021-02-13:
I can PR this in but I probably won't be able to test for a bit.

NoahBres, 2021-03-10:
Bluetooth PS4 Gamepad doesnt in either Firefox or Safari because they report different ID's

Bluetooth
 - Safari: DUALSHOCK 4 Wireless Controller Extended Gamepad
 - Firefox: 54c-9cc-DUALSHOCK 4 Wireless Controller
   - this doesn't work because the regex searches for 4 character VID/PID with the zero prepended

Wired
  - Safari: Wireless Controller Extended Gamepad
  - Firefox: 54c-9cc-Wireless Controller

NoahBres, 2021-03-10:
Other have reported that the F310 doesn't work with Dash. I'm assuming it's due to the button mapping changes or an ID change. I would like to test this but I believe it would be safe to just remove the gamepad specific mappings and default to the browser standardized one now.

NoahBres, 2021-03-10:
@TimVargo Would it be possible to go to https://gamepad-tester.com/ and record/report how the gamepad behaves?

It will look something like this:
https://user-images.githubusercontent.com/6739076/110693552-3006a800-81ad-11eb-8098-be2a8f4b3e9b.mp4

You can either just press all the buttons and record the screen or just post if the results match with the actual mappings. Take note of the button numbers.

Right now, in Dash, the Logitech's mappings are as follows:
  - A: B1
  - B: B2
  - X: B0
  - Y: B3
  - left stick x: axis 1
  - left stick y: axis 2
  - right stick x: axis 3
  - right stick y: axis 4
  - dpad up: 12
  - dpad down: 13
  - dpad left: 14
  - dpad right: 15

There are more mappings but you won't need it. The mappings are here if you're curious though: https://github.com/acmerobotics/ftc-dashboard/blob/4a7dd96f0873422e35eb1b4d11115f9b66c6edac/FtcDashboard/dash/src/store/middleware/gamepadMiddleware.ts#L85

The xbox and ps4 gamepad mappings are as follows:
  - A: B0
  - B: B1
  - X: B2
  - Y: B3
  -  left stick x: axis 0
  - left stick y: axis 1
  - right stick x: axis 2
  - right stick y: axis 3
  - dpad up: 12
  - dpad down: 13
  - dpad left: 14
  - dpad right: 15

I suspect what is happening is either the ID reporting at the top for the logitech is different or your mappings right now actually follow the Xbox/ps4 mappings (which is now the w3 standard).

NoahBres, 2021-03-10:
@smartynenko For PR #47, you tested the Xbox 360 mapping? 

Right stick X and Y are axes 3 and 4 which do not follow the w3 spec

https://github.com/acmerobotics/ftc-dashboard/blob/4a7dd96f0873422e35eb1b4d11115f9b66c6edac/FtcDashboard/dash/src/store/middleware/gamepadMiddleware.ts#L121

jasonappah, 2021-03-10:
Ngl I completely forgot about this but I can make the PR now and test tomorrow.

TimVargo, 2021-03-10:
Noah,
As you requested, my Logitech Gamepad F310 reports as "Xbox 360 Controller
(XInput STANDARD GAMEPAD)", and the mappings are indeed following the xbox
mappings as you suspected:
A: B0
B: B1
X: B2
Y: B3
left stick x: axis 0
left stick y: axis 1
right stick x: axis 2
right stick y: axis 3

    ~Tim Vargo

NoahBres, 2021-03-11:
@TimVargo Thank you so much!

@rbrott So is it okay to abolish the gamepad type completely and just use the standardized mappings?

rbrott, 2021-03-11:
I'm reluctant to change things again so soon. Maybe I'm missing something, but it seems like Tim's F310 should work with the current code given that it has an Xbox 360 ID?

smartynenko, 2021-03-11:
> @smartynenko For PR #47, you tested the Xbox 360 mapping?
> 
> Right stick X and Y are axes 3 and 4 which do not follow the w3 spec and do not match the PS4 mappings
> 
> https://github.com/acmerobotics/ftc-dashboard/blob/4a7dd96f0873422e35eb1b4d11115f9b66c6edac/FtcDashboard/dash/src/store/middleware/gamepadMiddleware.ts#L121

Sorry, must have checked in outdated version. I currently have

```
    case GamepadType.XBOX_360:
      return {
        left_stick_x: cleanMotionValues(gamepad.axes[0]),
        left_stick_y: -cleanMotionValues(-gamepad.axes[1]),
        right_stick_x: cleanMotionValues(gamepad.axes[2]),
        right_stick_y: -cleanMotionValues(-gamepad.axes[3]),
```

NoahBres, 2021-03-11:
> Sorry, must have checked in outdated version. I currently have

You're absolutely right. My apologies.

TimVargo, 2021-03-11:
To clarify what (doesn't) happen with F310, I plug in the gamepad and Windows does detect it, but pressing Start-A, or B, X, Y has no effect at all -- nothing happens with the dimmed gamepad icons.

smartynenko, 2021-03-11:
> To clarify what (doesn't) happen with F310, I plug in the gamepad and Windows does detect it, but pressing Start-A, or B, X, Y has no effect at all -- nothing happens with the dimmed gamepad icons.

Do you have it in X mode or D mode?

TimVargo, 2021-03-11:
The one I tested with yesterday has X mode is selected on underside.  Is 
that correct?

   ~Tim

jasonappah, 2021-03-16:
I was finally able to do a quick test using the default mapping using an F310 in xinput mode and all the buttons appear to be working except the guide button. I did have to do a bunch of Windows updates before I could get my browser to recognize the gamepads though.

jasonappah, 2021-04-11:
Update: Turns out I was a dum dum and used 0.4.0 for my testing instead of 0.4.1. In 0.4.1 the F310 seems to work fine without any modifications in xinput mode.

jasonappah, 2021-04-11:
Oops I was wrong, everything works except the right joystick. The Y-axis shows up as the X-axis, and the X-axis doesn't work at all, but still better than 0.4.0.

closed 2025-08-16 by rbrott