acmerobotics/ftc-dashboard #50: Changes Popup

Jarvis7529, 2021-03-02:
I was wondering if you can add a popup with here are the configs that changed since you started so you could make sure you update the real values in the code?

NoahBres, 2021-03-02:
I actually was thinking about this a while ago but don't really have an idea for how the UI would be implemented. Firefox's dev tools actually shows the modified css values, which was the impetus for the idea. However, I am not entirely sure how to translate that to Dashboard's UI.

Jarvis7529, 2021-03-05:
My thought is have a button up by the save disk that would popup a model window showing original and changed values. Maybe show all values with some JS that would allow you to show all or just changes,