acmerobotics/ftc-dashboard #59: [Feature Request]  Add keyboard gamepad emulation

12buntu, 2021-03-20:
Hey, I'm not sure if this is even technically possible, I haven't really looked at the code, but it would be really cool if the keyboard could emulate gamepad button presses when viewing FTC dashboard.

Sam948-byte, 2021-11-23:
I think it should be since all you would have to do is map some of the gamepad keys to the controller values.  I'm going to take a look and see what I can find.

wangxdflight, 2022-10-30:
That would be a very cool feature if it is available. Most of the time of my working with FTC SW, I just use an android phone and dashboard (to launch opModes). The only limitation I feel has been the lack of gamepad.
I thought of adding a customized gamepad implementation in FTC teamcode, and it can receive commands via network socket from another JAVA GUI app. It would be much better if gamepad inputs can be directly generated from buttons on the dashboard.