acmerobotics/ftc-dashboard #60: Dashboard stops working ( WebSocket connection to 'ws://192.168.43.1:8000/' failed)

EddieDL, 2021-04-18:
We have been using FTC-Dashboard for 2 years now without issue. Two days ago it just stopped working and we see the following error in the browser.

`WebSocket connection to 'ws://192.168.43.1:8000/' failed:`

We have tried to investigate the issue, but aren't sure where to look. We have also tried:
- Using versions of our code from before we had this issue
- Reinstalling the Control Hub OS (thinking maybe something was broken on the control hub).

What should we be doing to investigate why things stopped working.

[image]

EddieDL, 2021-04-21:
@rbrott @NoahBres Do either of you have any idea about why this would be happening?

EddieDL, 2021-04-24:
We figured this issue out. Apparently someone ran the "Disable Dashboard" command and no one else knew it. Since the web pages seems to be still sent back to the browser. Perhaps it should have a message stating that the dashboard is currently disabled.

Or you can just check for that :) 

Wish someone would have thought of that sooner.

closed 2021-04-24 by EddieDL

NoahBres, 2021-04-25:
@rbrott Can you kill the webserver as well when dash is disabled?

rbrott, 2021-04-25:
Sure, one can register `null` request handlers.

EddieDL, 2021-04-25:
Killing the webserver would also help, is that something you can do in the installation logic? Ideally just a way to know that it is turned off not "broken".