acmerobotics/ftc-dashboard #67: Dashboard breaks on Firefox - `navigator.getGamepads is not a function`

NoahBres, 2021-09-29:
User reported error on the Discord. Have yet to get around to confirming and reproducing myself.

Firefox version 92.0
Ubuntu version 21.04 (wayland)

[image]
[image]

NoahBres, 2021-09-30:
My current assumption—supported only via a cursory search—is that this issue relates to a relatively new API called [Secure contexts](https://www.w3.org/TR/secure-contexts/). This seems to be a relatively new W3C spec? It still holds a status of "Candidate Recommendation Draft." It has yet to even reach "W3C Recommendation" status.
However, Firefox did [commit to locking features down via secure contexts in 2018](https://blog.mozilla.org/security/2018/01/15/secure-contexts-everywhere/).

Secure contexts must be served over `https://` or `http://localhost` (via https://developer.mozilla.org/en-US/docs/Web/Security/Secure_Contexts)

It seems that the gamepad API has been locked down in Firefox relatively recently.
https://developer.mozilla.org/en-US/docs/Web/API/Navigator/getGamepads
[image]

https://caniuse.com/?search=gamepad.get
[image]

Basic gamepad functionality with the PS4 controller was tested both in dev and on the compiled app in Chrome, Safari, and Firefox. I don't believe this was an issue in the last PR touching gamepads.

More information: https://hacks.mozilla.org/2020/07/securing-gamepad-api/

[image]

This blog only indicates the use of secure contexts in the development versions. I am assuming that this was incorporated into the mainstream version of Firefox in version 91 if we reference the caniuse `getGamepads: Secure context required` chart.

This is my quick-uninformed-5-minute-search opinion on the problem. I could be completely off base here.

NoahBres, 2021-09-30:
Potential workaround if this is the issue (have yet to confirm on my end)

https://developer.mozilla.org/en-US/docs/Web/Security/Secure_Contexts#feature_detection

Do not run gamepad code if `window.isSecureContext` returns true and version is above or equal to Firefox 91. Alternatively, run a null check on `navigator.getGamepads()` and try catch?

Or perhaps serve dash over HTTPS?
https://stackoverflow.com/questions/31270613/https-server-on-android-device-using-nanohttpd/35765334

Not entirely sure of the implications via how dash hooks into NanoHTTPD.

NoahBres, 2021-09-30:
More discussion: 
https://github.com/w3c/gamepad/issues/145

This same behavior seems to be planned for Chrome as well?
https://news.thewindowsclub.com/google-chrome-now-blocks-gamepad-access-on-insecure-websites-101927/
https://www.chromestatus.com/feature/5138714634223616

Restriction is still disabled by default behind an experimental flag

NoahBres, 2021-09-30:
This comment confirms my original suspicion: https://github.com/w3c/gamepad/issues/145#issuecomment-899850875

NoahBres, 2021-09-30:
Both WebKit and Chromium seem to be committed to this same behavior, albeit release schedule unknown.

https://github.com/w3c/gamepad/issues/145#issuecomment-899850875
https://github.com/w3c/gamepad/pull/120

rbrott, 2021-09-30:
I don't want to fiddle with HTTPS. Let's guard the gamepad code with the secure context flag and refer users to the development instructions.

PR #72 (2022-01-11): Address getGamepads changes
Handle `navigator.getGamepads()` not being defined instead of crashing, and display a warning to the user. Fixes #67. I've tested in Chrome and Firefox. 

There is a `setTimeout` in `gamepadMiddleware.ts` that I'm not proud of, but it told me I couldn't dispatch while constructing the middleware. I've never touched React before, so I wouldn't know if there's a better way of achieving that ¯\_(ツ)_/¯
closed 2022-01-11 by rbrott