acmerobotics/ftc-dashboard #78: Can we make dashboard to use infrastructure WIFI instead of WIFI direct?

wangxdflight, 2022-02-23:
My android phone has version 10, and seems have stability in WIFI direct when running FTC robot controller. 
Sometimes the network info shows "unavailable", and I cannot connect to WIFI direct.

I am wondering if dashboard can run on the IP address assigned by WIFI infrastructure router, so that I can have reliable WIFI connection.

Thanks.

[image]

rbrott, 2022-02-23:
Yes, that's possible. Follow the development instructions [here](https://acmerobotics.github.io/ftc-dashboard/gettingstarted#development), making sure to input the desired IP.

wangxdflight, 2022-02-23:
What I really want is (student users can easily use) to use it like below, not involving more additional tools or steps.
192.168.1.10:8080/dash

I think we just need to add permissions in manifest.xml to accept all incoming HTTP non-secure connection request. Not sure if this is enough.

rbrott, 2022-02-24:
There's nothing in the dashboard code that requires WiFi direct. In fact, I believe most of the time that it's operating over SoftAP/full-blown CH AP. (But this is all outside my wheelhouse.)

Are you able to load the other pages served by the RC web server? Can you load the dash client? In the default configuration, the dash attempts to open the web socket connection using the same host as in the page URL.

wangxdflight, 2022-02-24:
You are right.... Actually it works well via AP router. 
It didn't work because it was stuck in a previous state. By turning off/on WIFI, this issue can be worked around. 
[image]

rbrott, 2022-02-24:
That is above my pay grade.

cc @NoahAndrews

closed 2022-05-10 by rbrott