acmerobotics/ftc-dashboard #79: include FtcDashboard-release.aar not working

wangxdflight, 2022-02-24:
I am trying to compile my own version of dashboard, and include it in build.dependencies.gradle as below.  But it doesn't work. Disable/enable dashboard is not shown in the menu, and web browser shows "Mime type unknown: uri='/dash' path='dash'".
Any help? Thanks.

    //implementation 'com.acmerobotics.dashboard:dashboard:0.4.4'
    implementation files('../libs/FtcDashboard-release.aar')

rbrott, 2022-02-24:
I haven't reproduced the issue, but I suspect you're missing some dependencies specified in the POM. You may need to add [these lines](https://github.com/acmerobotics/ftc-dashboard/blob/c27aeb71772b7a4f6400d8704da4d1d8d2fdb907/FtcDashboard/build.gradle#L98-L102).

wangxdflight, 2022-02-24:
Great...it works after adding these lines....It was weird it didn't complain in build process.

Appreciate your prompt help!

closed 2022-02-24 by wangxdflight