acmerobotics/ftc-dashboard #84: "Event Loop Detached" Error

Robocoder9000, 2022-09-05:
Hi! I'm pretty new to FTC Dashboard, and have been working through a few issues with FTC Dashboard. We're trying to run some sample calibration code on a Control Hub with an Android driver station. First of all, we can't connect the Android driver station at the same time as Dashboard- I'm unsure if that's normal due to the interfering communication or if that's an issue in of itself. As far as we're aware, we should be able to run op modes through FTC Dashboard so this shouldn't be an issue. However, the OP Mode section on Dashboard is showing the error "op mode detached", and is blank otherwise. Is there a known solution to this? Google isn't helping much here. Thanks for your help!

Cheers,

#16544 Stratobots

rbrott, 2022-09-05:
> First of all, we can't connect the Android driver station at the same time as Dashboard- I'm unsure if that's normal due to the interfering communication or if that's an issue in of itself. 

That is unusual. Dash should have little affect on normal Robocol communication aside from additional bandwidth consumption.

> However, the OP Mode section on Dashboard is showing the error "op mode detached", and is blank otherwise.

I'm not sure exactly what causes that issue. The RC app can be started in different ways, and I believe they affect the startup lifecycle. The only solution I know is killing and restarting the app. (You may need `adb` to do that properly on a Control Hub.)

Robocoder9000, 2022-09-05:
Thanks for the tips, we restarted both DS and Rc, updated some dependencies and the problem seems to be gone now.

closed 2022-09-06 by Robocoder9000