acmerobotics/ftc-dashboard #99: Stream camera view not with tensorflow

wangxdflight, 2022-10-30:
I am able to get camera view for opencv (apriltag) and vuforia, but not with tensor flow.
Is it supposed to work?

Thanks a lot!

rbrott, 2022-10-30:
It should work. To debug, you can try setting a breakpoint in `TFObjectDetectorImpl#accept()` and seeing if it gets called. You can also try using the DS camera stream preview which relies on many of the same code paths.

wangxdflight, 2022-10-30:
Thanks for your reply. It now works. But I had to do code change as below in ConceptTensorFlowObjectDetection.java:
         /*
          * Configure Vuforia by creating a Parameter object, and passing it to the Vuforia engine.
          */
-        VuforiaLocalizer.Parameters parameters = new VuforiaLocalizer.Parameters();
+        int cameraMonitorViewId = hardwareMap.appContext.getResources().getIdentifier("cameraMonitorViewId", "id", hardwareMap.appContext.getPackageName());
+
+        VuforiaLocalizer.Parameters parameters = new VuforiaLocalizer.Parameters(cameraMonitorViewId);

wangxdflight, 2022-10-30:
But the camera view in dashboard still doesn't show the detection bounding boxes.

rbrott, 2022-10-30:
If I'm reading the source right, more-or-less the same image is shown on the dashboard as is shown on the RC screen. You're free to write your own `CameraStreamSource` implementation if you want to further manipulate the `Bitmap` that will be sent.

wangxdflight, 2022-11-20:
It works now...Thanks.

closed 2022-11-20 by wangxdflight