acmerobotics/ftc-dashboard #104: Dashboard should have clear messaging indicating enabled vs. disabled state.

cmacfarl, 2022-11-18:
When the dashboard is disabled, it looks like it simply hasn't been initialized.  The opmode list is empty, there are no configuration items etc.

It would be useful to have either an overlay, or something in the blue banner indicating state, particularly when disabled so a user doesn't perhaps wonder why the dashboard appears broken.

NoahBres, 2022-11-18:
Doesn't disabling dashboard cut the server? So the client would not be able to indicate if it was disabled or if it simply wasn't connected?

cmacfarl, 2022-11-18:
It doesn't cut the server.  The server is serving OnBot Java, Blocks, and the Program and Manage page as well, so you can't just shut down the server.   So the dash route still returns a page.  What it does cut is the websocket connection, so you don't get the ping updates and such.  However, if you aren't intimately familiar with how the dashboard works it looks just like it's broken and you don't know why.   I didn't look into it much further than that, but it does appear that server itself knows that the dashboard is disabled, and hence presumably the route, that is still serving the page, could send an indication that the dashboard is disabled with a little explanatory text explaining how to turn it back on.

NoahBres, 2022-11-18:
My apologies. I mis-spoke confusing server and socket. 
The page is still served so perhaps it can serve an empty page with the appropriate indicator? Although I believe this would require a refresh? Or polling REST requests regarding status updates?
I'm sure Ryan has the proper solution. I am just trying to learn the workings.

cmacfarl, 2022-11-18:
I don't think it needs to be dynamic necessarily.   The natural thing to do is to refresh the page if things look wonky, at which point the disabled message could be displayed.

Although I do believe that the client should be able to know when the websocket is down and display a message appropriately.  Although I will admit to not having done any React development.

rbrott, 2022-11-18:
Cutting the websocket connection doesn't give much additional security. Cheating is still only a few changes away for determined teams. I'm leaning toward keeping the connection alive and displaying a proper help message when the client is disabled.

wuzihaoo, 2022-12-02:
> When the dashboard is disabled, it looks like it simply hasn't been initialized. The opmode list is empty, there are no configuration items etc.
>
> It would be useful to have either an overlay, or [...]

This is exactly my question: how to turn it back on if I disabled it? I tried disconnecting but still disabled.

rbrott, 2022-12-05:
Try running the op mode called "Enable/Disable Dashboard"

rbrott, 2022-12-05:
Also this is fixed in master.

closed 2022-12-05 by rbrott