acmerobotics/ftc-dashboard #124: Dashboard Lag

petergriffinnn, 2023-01-06:
Hey all, for some reason, when my team attempts to utilize the dashboard for roadrunner (We have 3 odo wheels) we move the robot, and there is about a 5 second delay between the actual robot moving and what's displayed on the dashboard. Anyone have any documented solutions?

rbrott, 2023-01-06:
Try upgrading? That's probably the cause if you're still on 0.4.5.

petergriffinnn, 2023-01-06:
I just change implementation 'com.acmerobotics.dashboard:dashboard:0.4.5' to the newest version? Where can I find the newest version? Best wishes

rbrott, 2023-01-07:
Releases are listed [here](https://github.com/acmerobotics/ftc-dashboard/releases). You can also be notified of new releases using the instructions [here](https://docs.github.com/en/account-and-profile/managing-subscriptions-and-notifications-on-github/setting-up-notifications/configuring-notifications#configuring-your-watch-settings-for-an-individual-repository).

petergriffinnn, 2023-01-18:
We updated the grade:
//
// build.gradle in TeamCode
//
// Most of the definitions for building your module reside in a common, shared
// file 'build.common.gradle'. Being factored in this way makes it easier to
// integrate updates to the FTC into your code. If you really need to customize
// the build definitions, you can place those customizations in this file, but
// please think carefully as to whether such customizations are really necessary
// before doing so.

// Custom definitions may go here

// Include common definitions from above.
apply from: '../build.common.gradle'
apply from: '../build.dependencies.gradle'

android {
    namespace = 'org.firstinspires.ftc.teamcode'
}

dependencies {
    implementation project(':FtcRobotController')
    annotationProcessor files('lib/OpModeAnnotationProcessor.jar')
    implementation 'org.apache.commons:commons-math3:3.6.1'

    implementation 'com.acmerobotics.roadrunner:core:0.5.5'
   // implementation 'com.acmerobotics.dashboard:dashboard:0.4.8'
}

dependencies {
    implementation 'org.openftc:easyopencv:1.5.2'
}
dependencies {
    implementation 'org.openftc:apriltag:1.1.0'
}

This still does not work. Any help?

qwertychouskie, 2023-01-18:
First, the relevant line is commented out.  Second, the line should be in `build.dependencies.gradle`

closed 2023-09-29 by rbrott