acmerobotics/ftc-dashboard #135: No connection icon black on dark-mode?

NoahBres, 2023-03-16:
[image]

Need to reproduce to confirm but this can be seen in #134's screenshots: https://github.com/acmerobotics/ftc-dashboard/pull/134#issuecomment-1445672807

rbrott, 2023-08-20:
Gamepad icons are black here. Not sure how I didn't notice this before.

This is Firefox 116.0.2 on macOS.

[image]

rbrott, 2025-04-06:
I think 08660ad965bfe4cd2df96e0e8e70dc7f7bb7f32e mostly resolves this

closed 2025-04-06 by rbrott