acmerobotics/ftc-dashboard #142: Graph not showing

PurplePizza123, 2023-10-07:
The graph is not showing while I am on the manual feedforward tuner. I went to the link to the FTC Dashboard. Next I went to ManualFeedforwardTuner Op Mode. Finally I started it and graphed the V0 and VRef. Finally I pressed the graph button in the top right and it did not graph anything.

[image]
[image]

Devildogs11206, 2023-10-07:
I have tried using 2 different laptops.

PurplePizza123, 2023-10-07:
I had to use microsoft edge instead of google chrome.

closed 2023-10-07 by PurplePizza123

qwertychouskie, 2023-10-07:
Chrome and Edge use the same engine (Chromium), so they should work the same.  Can you post the versions of Chrome and Edge, and whether you have any extensions installed?

Devildogs11206, 2023-10-07:
> Chrome and Edge use the same engine (Chromium), so they should work the same. Can you post the versions of Chrome and Edge, and whether you have any extensions installed?

Chrome version 117.0.5938.92 
Edge version 117.0.2045.47

PurplePizza123, 2023-10-07:
> Chrome and Edge use the same engine (Chromium), so they should work the same. Can you post the versions of Chrome and Edge, and whether you have any extensions installed?

I used 
Chrome version 117.0.5938.92
Edge version 117.0.2045.47
And no extensions