acmerobotics/ftc-dashboard #145: Add a list of object types that support editing?

Powercube7, 2023-11-09:
Hello there! I was wondering whether it would be feasible to add a list of object types that can be edited live by the dashboard? I've been confused quite a few times wondering whether a type was supported or not, and no information could be found in the documentation.

amarcolini, 2023-11-12:
Dashboard can modify the public non final fields of any object. It’s also recursive, so it can modify the fields of a field and so forth. Dashboard also has support for basic number, boolean, and string types (and more recently, arrays!)

Powercube7, 2023-11-12:
@amarcolini Thanks for the answer! Does it also support types such as Maps, HashMaps, Lists and Pairs?

amarcolini, 2023-11-12:
As far as I am aware, no.