acmerobotics/ftc-dashboard #147: Could not resolve FTC dashboard

Sirllamaa, 2023-12-08:
Anyone know whats going on? Tried reinstalling android studio for a gradle issue then this monstrosity popped up.

`Executing tasks: [:TeamCode:assembleDebug, :TeamCode:assembleDebugUnitTest, :TeamCode:assembleDebugAndroidTest] in project /Users/mpr070/Documents/GitHub/CenterStage2023-2024

> Configure project :FtcRobotController
AGPBI: {"kind":"warning","text":"The option setting 'android.defaults.buildfeatures.buildconfig=true' is deprecated.\nThe current default is 'false'.\nIt will be removed in version 9.0 of the Android Gradle plugin.\nYou can resolve this warning in Android Studio via `Refactor` > `Migrate BuildConfig to Gradle Build Files`","sources":[{}]}

> Task :TeamCode:createDebugVariantModel
> Task :TeamCode:preBuild UP-TO-DATE
> Task :TeamCode:preDebugBuild UP-TO-DATE
> Task :TeamCode:mergeDebugNativeDebugMetadata NO-SOURCE
> Task :FtcRobotController:preBuild UP-TO-DATE
> Task :FtcRobotController:preDebugBuild UP-TO-DATE
> Task :FtcRobotController:generateDebugBuildConfig
> Task :FtcRobotController:generateDebugResValues
> Task :FtcRobotController:generateDebugResources
> Task :FtcRobotController:packageDebugResources
> Task :FtcRobotController:javaPreCompileDebug
> Task :TeamCode:generateDebugBuildConfig
> Task :FtcRobotController:parseDebugLocalResources
> Task :TeamCode:javaPreCompileDebug
> Task :FtcRobotController:processDebugManifest
> Task :FtcRobotController:compileDebugLibraryResources
> Task :FtcRobotController:generateDebugRFile FAILED
> Task :FtcRobotController:writeDebugAarMetadata
> Task :TeamCode:generateDebugResValues
> Task :TeamCode:checkDebugAarMetadata FAILED
> Task :TeamCode:mapDebugSourceSetPaths FAILED
> Task :TeamCode:generateDebugResources
> Task :TeamCode:mergeDebugResources FAILED
> Task :TeamCode:createDebugCompatibleScreenManifests
> Task :FtcRobotController:extractDeepLinksDebug
> Task :TeamCode:extractDeepLinksDebug
> Task :TeamCode:processDebugMainManifest FAILED
> Task :FtcRobotController:mergeDebugShaders
> Task :FtcRobotController:compileDebugShaders NO-SOURCE
> Task :FtcRobotController:generateDebugAssets UP-TO-DATE
> Task :FtcRobotController:packageDebugAssets
> Task :TeamCode:mergeDebugShaders
> Task :TeamCode:compileDebugShaders NO-SOURCE
> Task :TeamCode:generateDebugAssets UP-TO-DATE
> Task :TeamCode:mergeDebugAssets FAILED
> Task :FtcRobotController:processDebugJavaRes NO-SOURCE
> Task :TeamCode:checkDebugDuplicateClasses FAILED
> Task :TeamCode:desugarDebugFileDependencies FAILED
> Task :FtcRobotController:mergeDebugJniLibFolders
> Task :FtcRobotController:mergeDebugNativeLibs NO-SOURCE
> Task :FtcRobotController:copyDebugJniLibsProjectOnly
> Task :TeamCode:mergeDebugJniLibFolders
> Task :TeamCode:mergeDebugNativeLibs FAILED
> Task :TeamCode:validateSigningDebug
> Task :TeamCode:writeDebugAppMetadata
> Task :TeamCode:writeDebugSigningConfigVersions
> Task :TeamCode:assembleDebugUnitTest UP-TO-DATE
> Task :TeamCode:preDebugAndroidTestBuild SKIPPED
> Task :TeamCode:processDebugAndroidTestManifest FAILED
> Task :TeamCode:javaPreCompileDebugAndroidTest
> Task :TeamCode:checkDebugAndroidTestAarMetadata FAILED
> Task :TeamCode:generateDebugAndroidTestResValues
> Task :TeamCode:mapDebugAndroidTestSourceSetPaths
> Task :TeamCode:generateDebugAndroidTestResources
> Task :TeamCode:mergeDebugAndroidTestResources
> Task :TeamCode:mergeDebugAndroidTestShaders
> Task :TeamCode:compileDebugAndroidTestShaders NO-SOURCE
> Task :TeamCode:generateDebugAndroidTestAssets UP-TO-DATE
> Task :TeamCode:mergeDebugAndroidTestAssets FAILED
> Task :TeamCode:processDebugAndroidTestJavaRes NO-SOURCE
> Task :TeamCode:checkDebugAndroidTestDuplicateClasses FAILED
> Task :TeamCode:mergeDebugAndroidTestJavaResource
> Task :TeamCode:desugarDebugAndroidTestFileDependencies
[... 367 lines cut ...]
         project :TeamCode > project :FtcRobotController
      > Could not resolve com.acmerobotics.dashboard:dashboard:0.4.14.
         > Could not get resource 'https://maven.brott.dev/com/acmerobotics/dashboard/dashboard/0.4.14/dashboard-0.4.14.pom'.
            > Could not GET 'https://maven.brott.dev/com/acmerobotics/dashboard/dashboard/0.4.14/dashboard-0.4.14.pom'.
               > The server may not support the client's requested TLS protocol versions: (TLSv1.2, TLSv1.3). You may need to configure the client to allow other protocols to be used. For more on this, please refer to https://docs.gradle.org/8.2/userguide/build_environment.html#sec:gradle_system_properties in the Gradle documentation.
                  > PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
   > Could not resolve com.acmerobotics.roadrunner:core:0.5.6.
     Required by:
         project :TeamCode
      > Could not resolve com.acmerobotics.roadrunner:core:0.5.6.
         > Could not get resource 'https://maven.brott.dev/com/acmerobotics/roadrunner/core/0.5.6/core-0.5.6.pom'.
            > Could not GET 'https://maven.brott.dev/com/acmerobotics/roadrunner/core/0.5.6/core-0.5.6.pom'.
               > The server may not support the client's requested TLS protocol versions: (TLSv1.2, TLSv1.3). You may need to configure the client to allow other protocols to be used. For more on this, please refer to https://docs.gradle.org/8.2/userguide/build_environment.html#sec:gradle_system_properties in the Gradle documentation.
                  > PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.
==============================================================================

BUILD FAILED in 1s
47 actionable tasks: 47 executed
`

qwertychouskie, 2023-12-09:
Looks like you aren't connected to the internet.  Far more than just the FTC Dashboard is failing to download in these error logs.

Sirllamaa, 2023-12-11:
I am connected to wifi though which is what i first thought it was so I tried at home too and with ethernet and nothing is working

Sirllamaa, 2023-12-11:
I have also built this before so it shouldn't even need to download

I probably should have included all that in the original issue

closed 2025-01-12 by rbrott