acmerobotics/ftc-dashboard #151: Centerstage PNG does not appear in Dashboard

do6453, 2024-01-27:
We are new to using the Dashboard software and did not mind only having the PowerPlay background. We have been using version 0.4.10 and now see that version 0.4.13 has the background we would like. FWIW, I tried to step thru each version of the Dashboard and code up with the following results:

**dashboard.0.4.11**
I/WebViewFactory: Loading com.android.webview version 52.0.2743.100 (code 275610050)
I/cr_LibraryLoader: Time to load native libraries: 3 ms (timestamps 9173-9176)
I/cr_LibraryLoader: Expected native library version number "52.0.2743.100", actual native library version number "52.0.2743.100"
W/asset: Asset path /data/app/com.qualcomm.ftcrobotcontroller-2/base.apk is neither a directory nor file (type=1).
E/ResourcesManager: failed to add asset path /data/app/com.qualcomm.ftcrobotcontroller-2/base.apk
D/AndroidRuntime: Shutting down VM
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qualcomm.ftcrobotcontroller, PID: 2940
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.qualcomm.ftcrobotcontroller/org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity}: android.view.InflateException: Binary XML file line #182: Binary XML file line #182: Error inflating class android.webkit.WebView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2666)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6121)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795)
     Caused by: android.view.InflateException: Binary XML file line #182: Binary XML file line #182: Error inflating class android.webkit.WebView
     Caused by: android.view.InflateException: Binary XML file line #182: Error inflating class android.webkit.WebView
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
        at android.view.LayoutInflater.createView(LayoutInflater.java:645)
        at com.android.internal.policy.PhoneLayoutInflater.onCreateView(PhoneLayoutInflater.java:58)
        at android.view.LayoutInflater.onCreateView(LayoutInflater.java:717)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:785)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:727)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:858)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:518)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:426)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:377)
        at com.android.internal.policy.PhoneWindow.setContentView(PhoneWindow.java:412)
        at android.app.Activity.setContentView(Activity.java:2428)
        at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:303)
        at android.app.Activity.performCreate(Activity.java:6709)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6121)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795)
     Caused by: android.content.res.Resources$NotFoundException: failed to load /system/app/webview/webview.apk
        at android.app.ResourcesManager.appendLibAssetForMainAssetPath(ResourcesManager.java:933)
        at android.webkit.WebViewDelegate.addWebViewAssetPath(WebViewDelegate.java:205)
        at com.android.webview.chromium.WebViewDelegateFactory$ProxyDelegate.addWebViewAssetPath(WebViewDelegateFactory.java:194)
        at com.android.webview.chromium.WebViewChromium.<init>(WebViewChromium.java:159)
        at com.android.webview.chromium.WebViewChromiumFactoryProvider.createWebView(WebViewChromiumFactoryProvider.java:446)
        at android.webkit.WebView.ensureProviderCreated(WebView.java:2336)
        at android.webkit.WebView.setOverScrollMode(WebView.java:2395)
E/AndroidRuntime:     at android.view.View.<init>(View.java:4023)
        at android.view.View.<init>(View.java:4146)
        at android.view.ViewGroup.<init>(ViewGroup.java:579)
        at android.widget.AbsoluteLayout.<init>(AbsoluteLayout.java:55)
        at android.webkit.WebView.<init>(WebView.java:627)
        at android.webkit.WebView.<init>(WebView.java:572)
        at android.webkit.WebView.<init>(WebView.java:555)
        at android.webkit.WebView.<init>(WebView.java:542)
        	... 27 more
Install successfully finished in 13 s 187 ms.
[... 331 lines cut ...]
I/FTCService: onNetworkConnectionEvent: CONNECTION_INFO_AVAILABLE
I/RobotCore: password
I/FTCService: Network Connection Passphrase: password
V/CoreRobotWebServer: starting port=8080
V/RobotCore: Getting network assistant of type: RCWIRELESSAP
V/CoreRobotWebServer: started port=8080
V/FTCService: RCActivity.controllerService=bound
V/UpdateUI: Network: active, disconnected
V/RobotCore: Robot Status: stopped
V/TooTallWebSocketServer: Started WebSocket server on port 8081
V/ThreadPool: container(0x0543cef1: default threadpool) added id=180 TID=0 count=1
V/RobotCore: thread: 'RobotSetupRunnable.run()' starting...
V/FTCService: Processing robot setup
V/RobotCore: Robot Status: stopped, scanning for USB devices
I/AppAliveNotifier: Telling the OS to set the RC alive notification timeout to 10 seconds
V/WebSocketManager: Registered handler for namespace ONBOTJAVA
V/WebSocketManager: Registered handler for namespace MC
I/[Gralloc]:  ion device open success! fd = 92
V/AppUtil: rootActivity=PermissionValidatorWrapper destroyed
V/AppUtil: rootActivity=FtcRobotControllerActivity
I/PermissionValidatorActivity: Lifecycle onDestroy : 21fe8bb
Install successfully finished in 19 s 172 ms.
$ adb shell am start -n "com.qualcomm.ftcrobotcontroller/org.firstinspires.ftc.robotcontroller.internal.PermissionValidatorWrapper" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER

The **only** thing I have changed between each push to my robot was the version of the Dashboard.

do6453, 2024-01-27:
Here is a colored version of the file, if that helps.
[Dashboard.docx](https://github.com/acmerobotics/ftc-dashboard/files/14074264/Dashboard.docx)

rbrott, 2024-01-27:
As of version v0.4.11, the field is no longer there all the time. It only appears when you run an op mode and start drawing things on it. So I would try proceeding with v0.4.15 and seeing if the new field pops up when you run your programs.

closed 2025-01-12 by rbrott