acmerobotics/ftc-dashboard #153: [Feature] Possibly add a way to display User Code Exceptions on the Dashboard?

GraysonTinker, 2024-07-18:
Currently, unlike global errors, User Code Exceptions aren't displayed in any way on the dashboard. This isn't necessarily bad, as the errors are still visible in logcat, but it can be a bit unintuitive to have an opmode suddenly stop without any indication as to why, especially when it occurs as a result of hardwareMap errors during the init phase.

While investigating the possibility of making a PR for this issue, I looked into how Exception data originating from user code travels. The data that is currently acquired and sent to the dashboard comes from `RobotLog.getGlobalErrorMessage()`, which is set by `RobotLog.setGlobalErrorMessage()`. However, User Code Exceptions are handled differently: instead, they are simply logged as an error, and have up to 15 lines sent to the Driver Hub via the `NetworkConnectionHandler`. This occurs inside `OpModeManagerImpl.handleUserCodeException()`.

Here are some potential ways to add this functionality to the Dashboard if this seems like something worth pursuing:
1. Modify some part of `RobotCore` to send the Exception to us, potentially raising questions about whether this is possible to do, desirable in the first place, and/or legal.
2. Attempt to intercept the CMD_SHOW_STACKTRACE Command sent to the Driver Station, and the associated stack trace data.
    - An idea on how this could be done is emulating a Driver Station network connection to localhost. I don't know how realistic that would actually be though.
3. Attempt to read the system log and parse out errors.
    - It seems like a consistent and unique tag is used, "OpModeManager", so it might not be too hard to filter.
    - This could potentially be expanded into a new Logcat Dashboard element, which might be useful in and of itself.
4. Don't actually obtain Exception data, but instead display a preconfigured message saying the code crashed and pointing the user to logcat.
    - Potentially this could be done by checking if the opmode lasted less than a certain amount of time, though it wouldn't catch all cases.
5. Make no changes and leave it how it is currently. It does work fine once you get used to it.

Thanks for reading

rbrott, 2024-07-22:
I agree that it would be nice to show user exceptions as well. Kudos for tracing how the FTC SDK handles those exceptions. If you're interested in pursuing this further, I would open an issue against the SDK repo asking why user exceptions aren't incorporated into the global error message and then dash will automatically get this feature without additional effort.

GraysonTinker, 2024-07-22:
Got it, will ask.

NoahAndrews, 2024-07-23:
> If you're interested in pursuing this further, I would open an issue against the SDK repo asking why user exceptions aren't incorporated into the global error message and then dash will automatically get this feature without additional effort.

We do it that way to enable a much better user experience on the Driver Station.

> Attempt to intercept the CMD_SHOW_STACKTRACE Command sent to the Driver Station, and the associated stack trace data.

I actually wrote some code 6 years ago that adds an API for registering a callback for every command that gets sent to the DS. I can see about resurrecting it for 9.0.