acmerobotics/ftc-dashboard #155: Custom Field Image Not Continuously Displaying On Dashboard

Aldhanekaa, 2024-09-16:
Hello all, as described on the Official FTC Dashboard documentation [here](https://acmerobotics.github.io/ftc-dashboard/fieldview) and https://github.com/acmerobotics/ftc-dashboard/issues/81 , it's possible to override default field drawing with custom field image by implementing `packet.fieldOverlay().drawImage()`

I've implemented as described on the docs and tested it, it works, however it only shows once and further after that it won't show the custom field image but instead the default field image. I also notice that my custom field image is overlaying the robot circle on the field. 

Is there any solution for this? 

https://github.com/user-attachments/assets/f5e813fe-3e21-4144-aabd-496627e1a841

here is my following code : 

```java
TelemetryPacket packet = new TelemetryPacket(false);
        packet.fieldOverlay()
                .drawImage("/dash/FGC_FIELD.jpg", 0, 0, 144, 144);

        SampleTankDrive drive = new SampleTankDrive(hardwareMap, telemetry);
        drive.setMode(DcMotor.RunMode.RUN_WITHOUT_ENCODER);
        FtcDashboard dashboard = FtcDashboard.getInstance();

        waitForStart();

        packet.fieldOverlay()
                .drawImage("/dash/FGC_FIELD.jpg", 0, 0, 144, 144);
        while (!isStopRequested()) {
            dashboard.sendTelemetryPacket(packet);
```

rbrott, 2024-09-17:
Is that all of your code? I suspect something is sending telemetry to dash without overriding the image.

rbrott, 2024-09-17:
Yes unfortunately that other code does matter. Internally that's sending another packet with the default field.

rbrott, 2024-09-17:
Actually can you try putting
```
TelemetryPacket packet = new TelemetryPacket(false);
        packet.fieldOverlay()
                .drawImage("/dash/FGC_FIELD.jpg", 0, 0, 144, 144);
```
inside the loop? Each packet should only be sent once.

closed 2025-01-12 by rbrott