acmerobotics/ftc-dashboard #157: I am trying to run my auto mode and I get an error only after I press stop. Nothing happens when I press start and when I press stop this error occurs

Leoshrotri, 2024-10-24:
I have no idea why this is popping up because I already have my control hub, expansion hub, and driver hub all up to date on the 1.8.2 version that it is suggesting. I tried REV Hardware Client or whatever its called and everything is up to date. I have no idea why no code will even run and why the error only shows up after I click stop. I am pretty sure there is no problem with the coding but it is attatched below along with the error.
[image]
[image]
[image]
[image]
[image]

TuckerzKnight, 2024-10-30:
Sorry for not replying earlier! I'm not too familiar with Roadrunner, so correct me if I'm wrong... In your 4th picture at line 36, you called runOpMode(). I don't think you need to call the function, I believe it just runs when you hit play on the driver station. Also just out of curiosity, is there supposed to be any code contained in initTfod()?
Are you using anything like LynxFirmware.throwIfModulesAreOutdated(hardwareMap); in your code? Maybe thats something to look into. IDK, good luck

Leoshrotri, 2024-10-30:
Thank you, this did fix the issue. You helped a lot.

closed 2025-01-12 by rbrott