acmerobotics/ftc-dashboard #166: Saving telemetry on the robot

meejah, 2025-03-05:
Hello,

Thanks for everyone's work on FTCDashboard, it is a great tool and we love it! It is definitely on my short list of "essential FTC libraries".

It would be great to be able to save telemetry logs on the robot -- for example, during competitions when sending telemetry to an external source is against the rules. Is there already a way to do this?

If there is not, would you accept a PR to make that easy -- if so, I will attempt to make my programmers excited about producing such a PR :) The idea would be to serialize "a JSON object per line" to a file on the controller, such that it can be pulled off later.

rbrott, 2025-03-08:
> It would be great to be able to save telemetry logs on the robot -- for example, during competitions when sending telemetry to an external source is against the rules. Is there already a way to do this?

There's a logging view that may do what you want: https://github.com/acmerobotics/ftc-dashboard/pull/55. I'd be curious if that meets your needs.

Glad you find the library useful :)

meejah, 2025-03-11:
We also have https://github.com/meejah/blckbx which can save all telemetry to a JSON file.

What we want is as easy way to write each `TelemetryPacket` as single-line JSON to a file on the robot. So like both of the above solutions, but without WebSockets etc involved so that it is competition legal (and we'd pull the logs off after a match with `adb` or similar).

rbrott, 2025-03-17:
Okay that functionality does not exist in the current library to my knowledge. I'm open to including it (possibly as a step toward replaying matches in dash), though I'd want it to be implemented in a way that doesn't increase loop times or fill up the disk (saving every telemetry packet for a long match as JSON may be a lot of space). It would be nice if it were opt-in.

meejah, 2025-03-17:
Yes, okay those all sound like good things -- definitely don't want a robot crash because of telemetry problems! I'll discuss with my team and if we get a solution that works for us, we'll propose an upstream PR (or even just an example, depending).

Thanks for the feedback!

(replaying in the dashboard would indeed be cool too!)