acmerobotics/ftc-dashboard #184: Dashboard plugins

DABANANAMAN120, 2025-08-04:
Is there any way to allow users to create plugins for the custom view? If not, would you consider adding one? This not only would allow you to split the load among the community, but also have more creativity in the development process. For example, one of the ideas I had was having a physical robot layout similar to the photo seen at https://frcdashboard.github.io/ , which would really only be possible if users could either upload diagrams and drag telemetry data onto it OR have the users create the widget for their bot from scratch. Just some food for thought. Thanks for this great tool!

rbrott, 2025-08-07:
The best way to add custom views to dash is to fork it IMO and modify the code appropriately. I've considered making an API for plugins but it's not clear to me what's the right way to structure it; many interesting plugins will want to modify both the frontend and the backend. I'm open to suggestions on this front.

DABANANAMAN120, 2025-08-16:
@rbrott it's not by any means a top priority, but I think that creating such an API would be an interesting path to look into. I mainly work with python, not so much with Java, so I may not be able to give that much assistance, but I would be willing to help out with developing this if you'd like!