acmerobotics/ftc-dashboard #185: Various ClassNotFound errors when using dashboard

polarbub, 2025-08-05:
When I add the dashboard to a otherwise stock [FtcRobotController](https://github.com/FIRST-Tech-Challenge/FtcRobotController) it causes a bunch of ClassNotFound errors in logcat. It doesn't seem to effect functionality. Is this something to be worried about? The errors are attached.

I added the dashboard by modifiying the bottom of `TeamCode/build.gradle` to have
```repositories {
    maven {
       url = 'https://maven.brott.dev/'
    }
}

dependencies {
    implementation project(':FtcRobotController')
    implementation "com.acmerobotics.dashboard:dashboard:0.4.17"
}```

[errors.txt](https://github.com/user-attachments/files/21606387/errors.txt)

rbrott, 2025-08-07:
I wouldn't worry about those errors. It's expected that some classes can't be resolved and dash will ignore them for config purposes: https://github.com/acmerobotics/ftc-dashboard/blob/6fc8cf87fbcb4fb5b7894065c229c3d1bb0029cf/FtcDashboard/src/main/java/com/acmerobotics/dashboard/FtcDashboard.java#L740-L743. As far as I can tell the logcat messages are coming from ART itself and aren't easily suppressed.

polarbub, 2025-08-07:
Okay. Thanks for the quick reply!